Richard-Gemmell/teensy4_i2c #1: I2C trouble

philippe488, 2019-12-11:
Hello, 
Moving from teensy LC to teensy 4.0 
I tranfert I2C comm from i2c_t3 to your brilliant I2C

//#include <i2c_t3.h>

#include <Arduino.h>
#include <i2c_driver.h>
#include "imx_rt1060/imx_rt1060_i2c_driver.h"

#include "FastLED.h"

The transfert with a Raspberry 4.0 at 1Hz around 100 write and 50 read. I tried writing by burst up to byte per byte with delay beetwen them

Some octets are not detected by Teensyor some octets are added with Zero content.
This is detected in main loop checking databufI2Cr content.
Did I miss Something ?

Regards

here part of my code

//#define MEM_I2C_LENW 64
#define MEM_I2C_LENW 128
volatile byte databufI2Cw_i[MEM_I2C_LENW];
volatile byte databufI2Cw[MEM_I2C_LENW];
//#define MEM_I2C_LENR 128
#define MEM_I2C_LENR 256
volatile byte  databufI2Cr_i[MEM_I2C_LENR];
volatile byte  databufI2Cr[MEM_I2C_LENR];
//volatile uint8_t receivedI2C,sendI2C;
volatile unsigned short receivedI2C,sendI2C;

void receiveEvent( int count)
{
 // This is the only time we can guarantee that the
// receive buffer is not changing.
// Copy the content so we can handle it in the main loop.

  memcpy((byte*) &(databufI2Cr[receivedI2C]), databufI2Cr_i, count);
  receivedI2C +=count;   
}

in setup
slave.after_receive(receiveEvent);
  slave.set_receive_buffer(databufI2Cr_i, MEM_I2C_LENR);
  slave.before_transmit(requestEvent);
  slave.after_transmit(requestEvent_end);
  slave.set_transmit_buffer(databufI2Cw_i, MEM_I2C_LENW);

  // Enable the slave
  slave.listen(teensy_I2C_ADDRESS);

Thanks you in advance

Philippe

Richard-Gemmell, 2019-12-11:
Hi Philippe,

This sounds like an electrical problem. I'm using external pull-up resistors between the Teensy and the Raspberry Pi. I'm currently using 1k Ohm but 2.2k should work too. There's one for SDA and another for SCL. If you haven't already done so, you should connect the GND pins of the Raspberry Pi and the Teensy as well.

Unfortunately, I still get occasional errors. It varies between 1 error per 100,000 bytes and 1 per million bytes.

I can't see anything wrong with the code snippet apart from the potential buffer overrun in receiveEvent() if receivedI2C gets too large. I doubt that's the problem though.

cheers,
Richard

philippe488, 2019-12-11:
Thanks for quick and valuable answer, since my soft look good.
I had in mind that i could be a resistor issue.
In fact the board are connected through differential i2c sandbox pca9600.
I just checked that the pull up are 5kohm.
For the raspberry i have to check the pull up of other i2c devices for teensy it is the only device so it it 5kohms. 
I understand this is too much...is so i will add a

philippe488, 2019-12-11:
Resistor and oet you know.
Regards
Philippe

philippe488, 2019-12-11:
You are the best !

I added a 2k resistor (+5K) in // that makes 1.42 kOhm. in the Teensy side. I remind you that I have a differential interface in addition. No more error !

But...there is always a but in electronic developpement. 
The fast led Library fastled.show interfere with I2C/ I still have many I2C errors when sending data to leds. In teensy LC, the led driver was releasing the interruption...not sure it does with teensy 4: The rapsberry driver says time-out write

However many thanks, your Library is working good and is simple to use

Philippe
(from France)

closed 2019-12-11 by philippe488

Richard-Gemmell, 2019-12-11:
I'm glad to hear that's working. Thanks for letting me know. 

I'm glad you found it easy to use.

cheers,
Richard