Richard-Gemmell/teensy4_i2c #2: Slave not detected by STM32F103

threeR3, 2020-01-23:
I am trying to use a STM32F103 as a master for a Teensy 4.0 slave. I'm using https://github.com/rogerclarkmelbourne/Arduino_STM32 for the STM32, and your library for the Teensy. I can see the Teensy via a I2C scan from an arduino nano, but not from the STM32. The STM32 can see other hardware I2c devices fine. Could there be something different in the I2C handling that would cause issues? It's odd that I can see the Teensy from the Nano, but not the STM32.

I am using i2c_register_slave for the teensy and a general purpose scanner (that picks up other i2c devices ok) on the STM32.

Richard-Gemmell, 2020-01-23:
It's possible that the method that Arduino_STM32 uses to scan for devices doesn't work with teensy4_i2c. If this is the case then you should be able to read the Teensy even if you can't pick it up with a scan. If you post the code that you're using to do the scan from the STM I'll check it out and see if it works.

It's possible that it's just an electrical problem. The slave and master need to have a common ground. The Teensy also needs external pullup resistors. (Try 1k or 2.2k.) Different devices contribute different amounts of pullup resistance so it's possible that one pair of devices works when another pair don't. (I apologise if you've already done this.)

cheers,
Richard

threeR3, 2020-01-23:
Hey Thanks for the quick response.

Here's the scan code on the STM32:

`// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//

/* Example pinmap for Bluepill I2Cs (by Testato)
 I2C-1 standard pins: PB7(sda) PB6(scl)
 Use it by "Wire" without pin declaration
  Wire.begin();
 I2C-1 alternative pins: PB9(sda) PB8(scl)
 Remap the first I2C before call begin()
  Wire.setSDA(PB9);
  Wire.setSCL(PB8);
  Wire.begin();
 I2C-2: PB11(sda) PB10(scl)
 Remap the second I2C before call begin()
  Wire.setSDA(PB11);
  Wire.setSCL(PB10);
  Wire.begin();
 If you want to use the two I2Cs simultaneously, create a new instance for the second I2C
  TwoWire Wire2(PB11,PB10);
  Wire2.begin();
 
*/

#include <Wire.h>

void setup() {

  Serial.begin(9600);
  Wire.begin();
  Wire.setClock(400000);
  Serial.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.

    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);

      nDevices++;
    }
    else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("done");

  delay(5000);           // wait 5 seconds for next scan
}`

I have pullups to 3.3v on both I2C lines, 2.2k at the moment. I've also tried 4.7k. I'll have to get some 1k to try and would be genuinely surprised if that was the kicker. I am also connecting the ground of both devices. 

A bit more niche, but the end game is to talk to the Teensy from a Zynq 7k. Neither the STM32 or Zynq works.. and the STM32 is much more common and easier to work with. I'm hoping whatever the issue is with the STM32, the zynq will follow suit.

Richard-Gemmell, 2020-01-23:
I'm pretty sure that code should find the Teensy Ok. The fact that it works from the Nano suggests there's nothing fundamentally wrong in teensy4_i2c.

I'm pretty sure you'll have done this already but it's worth double checking that the pins you're using match the instance of Wire or Slave at each end. If you've connected the Nano to one I2C port on the Teensy and the STM to the other then the I2CRegisterSlave() constructor will take Slave for one connection and Slave1 for the other. I admin I'm clutching at straws here. :)

2.2k should be enough but I wouldn't expect 4.7k to work well.

cheers,
Richard

threeR3, 2020-01-24:
Okay so I got it working in the sense the device is visable. Frankly I'm not sure what the particular change was that fixed it. I'm almost certain it was something with the hardware setup. Perhaps my breadboard is flaky. The issue I'm having now is the data being read back from the master periodically (like one in every 5) looks like garbage. I need to get some lower k resistors and see if that helps as this seems like a noisy electrical problem. I'll go ahead and close this as it doesn't appear to be related to the STM. Thanks for your help.

closed 2020-01-24 by threeR3

Richard-Gemmell, 2020-01-24:
I'm glad to hear you're making progress. Thanks for letting me know.