Richard-Gemmell/teensy4_i2c #3: Wire.requestFrom always returns 0

btashton, 2020-02-17:
The API indicates that this should return the number of bytes read, instead we always get 0 back.

https://www.arduino.cc/en/Reference/WireRequestFrom
```
Returns

byte : the number of bytes returned from the slave device 
```

This libraries implementation 

```
uint8_t I2CDriverWire::requestFrom(int address, int quantity, int stop) {
    rx_bytes_available = quantity;
    rx_next_byte_to_read = 0;
    master.read_async(address, rxBuffer, min((size_t)quantity, rx_buffer_length), stop);
    finish();
    return 0;
}
```

Richard-Gemmell, 2020-02-17:
Hi Brennan,
Thanks for pointing that out. I'll improve it.

Out of curiosity, do you have a requirement to know how many bytes were read if the read request fails part way through?

cheers,
Richard

Richard-Gemmell, 2020-02-17:
I've fixed it. It now reports the actual number of bytes received. This should be correct even if the transfer is aborted part way through. This also ensures that available() will return the correct value if the request fails. Please give a go and let me  know if it works for you.

cheers,
Richard

btashton, 2020-02-17:
Thanks for the quick fix.  I came across this because of a library I was pulling in that used Adafruid_BusIO, I updated the libraries to use the Wire implementation you wrote, but I ran into this issue here: https://github.com/btashton/Adafruit_BusIO/blob/31298baae0784eea07b47c3a7d19efbf996b2bd5/Adafruit_I2CDevice.cpp#L160

I had worked around this quickly by just returning the requested length, and it worked, so I suspect your fix will work even better.

Richard-Gemmell, 2020-02-17:
Ok. Thanks for letting me know.

btashton, 2020-02-17:
Just tested and it works great.  Thanks.

closed 2020-02-17 by btashton