Richard-Gemmell/teensy4_i2c #4: Teensy 4.0 not working in slave mode

Iambecomeroot, 2020-02-18:
I am trying to use the Teensy as an I2C slave with a Raspberry Pi. I flashed the [example code](https://github.com/Richard-Gemmell/teensy4_i2c/blob/master/examples/simple/i2c_register_slave/i2c_register_slave.ino) onto it. The LED is flashing as expected.

When I run `sudo i2c-detect -y 1` on the pi, I don't see the slave address. When I try to communicate with the Teensy via the `smbus2` library, I get a `Remote I/O error`.

I tried with both I2C ports of the Teensy (SDA: 18, SCL 19 and SDA: 17, SCL: 16) because it is not clear which port this library uses or how to change it. I also tried with external 1k pullup resistors.

I used a polulu IMU on the same bus to verify that it is not a problem on the side of the Pi and I can read it's address just fine.

Here is my short python code:
```
from smbus2 import SMBus
bus = SMBus(1)
bus.write_i2c_block_data(0x2d, 0x27, list(range(32)))
```

Any guidance would be appreciated.

Richard-Gemmell, 2020-02-18:
Hi Marcel,

If `i2cdetect -y 1`  doesn't show anything then there's probably a problem with the wiring. You definitely need the external 1k pullups. The Pi and the Teensy must also have a common GND. That example code uses port 0 on the Teensy. (Pins 18 and 19)

If you want to use port 1 then you need to change line 37 to refer to Slave1 instead of Slave in the I2CRegisterSlave constructor. I'll tweak the readme to make it a bit more obvious which ports match which object.

Your Python code won't work because it's trying to write too much data to an invalid register. I'd start out by trying to read a register like this:-

`bus = SMBus(1)`
`print(bus.read_i2c_block_data(0x2d, 0xA, 4))`

I've just tried it on my Pi. It prints `[212, 48, 0, 0]`.

cheers,
Richard

Iambecomeroot, 2020-02-18:
Hi Richard,

When I saw your reply, I came back to my setup and `i2c-detect -y 1` worked without changing anything. Maybe a lose cable, so a hardware problem, as you say.

You are quite right about sending too much data. I have merged `raw_slave_receiver` and `raw_simple_sensor` and everything is working perfectly. Thank you for this fantastic library!

Cheers,
Marcel

closed 2020-02-18 by Iambecomeroot