Richard-Gemmell/teensy4_i2c #6: Wire.Write Error

weirdwolf, 2020-05-14:
Hi there, 

I'm currently working on a smaller project in which i need to use the Teensy 4 as an I2C slave, aswell as needing the Teensy Audio library, and got referred to your library by someone on the Teensy forums.

In order to use the Audio library and your I2C library i've changed over the audio libraries Wire dependencies to yours, aswell as including "i2c_driver.h" and "i2c_driver_wire.h" in my sketch, but i'm getting an error that, as far as i can tell, is telling me that Wire.write() is being defined in two places.

Wire usage in my sketch is very simple, and is just set up to run a function when it receives a request event ( so Wire.begin(addr); & Wire.requestEvent(function); ) 

the function looks like so:
`void SendFFT() {
  Serial.println("I2C Requested");
  for(int i = 0; i > 9; i++){
      Wire1.write(levelI[i]);
  } 
  Serial.println("I2C Sent"); 
} `

and the error looks like so: 
`Arduino: 1.8.11 (Windows 10), TD: 1.50, Board: "Teensy 4.0, Serial, 600 MHz, Faster, US English"

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio\control_tlv320aic3206.cpp: In member function 'bool AudioControlTLV320AIC3206::aic_goToPage(byte)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio\control_tlv320aic3206.cpp:561:18: error: call of overloaded 'write(int)' is ambiguous

   Wire.write(0x00); delay(10);// page register  //was delay(10) from BPF

                  ^

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio\control_tlv320aic3206.cpp:11:0:

C:\Users\Weirdwolf\Documents\Arduino\libraries\teensy4_i2c\src/i2c_driver_wire.h:52:12: note: candidate: virtual size_t I2CDriverWire::write(uint8_t)

     size_t write(uint8_t data) override;

            ^

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/Stream.h:24:0,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/HardwareSerial.h:106,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/WProgram.h:46,

                 from C:\Users\WEIRDW~1\AppData\Local\Temp\arduino_build_875490/pch/Arduino.h:6,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio\control_tlv320aic3206.h:58,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio\control_tlv320aic3206.cpp:10:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/Print.h:59:9: note: candidate: size_t Print::write(const char*)

  size_t write(const char *str)   { return write((const uint8_t *)str, strlen(str)); }

         ^
`

I look forward to any advice & help you can give me, 
Regards,
Weirdwolf.

weirdwolf, 2020-05-14:
Quick Update: Hbit on the Teensy forum got me a workaround for the issue, by simply commenting out the dependency for "control_tlv320aic3206.h" (and removing it from the folder it was in) in the Audio.h library, as a result my code is now compiling and running (still have issues, but i believe they're unrelated)

Richard-Gemmell, 2020-05-15:
Hi weirdwolf,
The root cause of the problem was that control_tlv320aic3206.cpp contained a #include for Wire.h. That brought in the original Wire implementation which conflicted with my version. Hbit's fix worked by removing control_tlv320aic3206.cpp completely but you could also have fixed it by changing control_tlv320aic3206.cpp to include i2c_driver_wire.h instead of Wire.h. Unfortunately, there's no way for my library to do this automatically.

See https://github.com/Richard-Gemmell/teensy4_i2c#replacing-wireh for details.

cheers,
Richard

weirdwolf, 2020-05-15:
I already did that, changing tlv320 to include your i2c_driver_wire.h instead of Wire.h i mean, which is why i posted the issue here, I removed the Wire.h library entirely from the folder (tempoarily), and started compiling away to find every file i needed to change over to i2c_driver_wire.h, after i was no longer getting told Wire.h was missing, the error above popped up

Richard-Gemmell, 2020-05-15:
I'm sorry. I misunderstood the bug report. Thanks for the clarification.

I'll see if I can reproduce the problem. If you've got a very simple sketch that triggers it then please post it.

weirdwolf, 2020-05-15:
No problem at all.
I've attached the sketch I'm currently working on, that i know triggers the error, for me at least.
[SvendeProjekt_Teensy_FFT.zip](https://github.com/Richard-Gemmell/teensy4_i2c/files/4633684/SvendeProjekt_Teensy_FFT.zip)

Richard-Gemmell, 2020-05-15:
Thanks. I'll start digging.

Richard-Gemmell, 2020-05-15:
Hi weirdwolf,

I've released a new version which will prevent the compilation error you were seeing. The problem was that the original version of Wire contained an override for write() which I didn't have. This left the compiler unable to work out which of the remaining versions of write() it should call. I've fixed this by adding the missing override to I2CDriverWire.

`inline size_t write(int n) { return write((uint8_t)n); }`

I've also improved the documentation to make it clear that you need to replace all usages of Wire.h in any library dependencies and not just in the headers that you include. In your case, this means replacing the `#include "Wire.h"` in each of the control_xxx.cpp files in your hardware\teensy\avr\libraries\Audio directory.

I believe you can now restore all the files you deleted. Just changing the #includes in the Audio directory should be sufficient to make it work.

Please try it and let me know if this has solved the problem for you.

cheers,
Richard

weirdwolf, 2020-05-16:
Absolutely perfect mate, just updated the library and tested it out with the files restored and re-included, compiles with no issues, Thanks a ton for the quick work! 

I'll go ahead and close the issue since the problem has been resolved.

Cheers, and thanks again.

closed 2020-05-16 by weirdwolf