Richard-Gemmell/teensy4_i2c #8: Teensy i2c to RPi, i2c crashing, needs reset

mano1979, 2020-07-17:
Hi,

I have a teensy that i want to send multiple values to the RPi. However,...

The example sketches works most of the time and sometimes stops working, giving an error on the RPi: 
`pi@raspberrypi:~ $ python3 i2ctest.py
Traceback (most recent call last):
  File "i2ctest.py", line 14, in <module>
    block = bus.read_i2c_block_data(8, 0, 6)
  File "/home/pi/.local/lib/python3.7/site-packages/smbus2/smbus2.py", line 596, in read_i2c_block_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 5] Input/output error`

But when i implement this example in my existing code (also using a SPI RF24L01 radio module with the RF24 library) the i2c connection doesn't even show up when i do a scan op the Pi.

It used to work for a short few messages, but then it crashed with the same message as above. but instead of "[errno 5]" it said "[errno 121]".

The i2c connection can be found in a scan until the radio starts working, leaving me to wonder if they maybe use the same timer or so? because the radio doesn't start working until the i2c connection crashed.

I have tried all i could find online, from pullups to twisting wires and holding them between my fingers. I'm ready to start using voodoo if i thought that would help.

if you need my code, please say so. I'd be happy to share it. allthough i think the sketches are not the problem...

Richard-Gemmell, 2020-07-18:
Hi Mano,

If I've understood you correctly, you've got 2 problems. Firstly, the I2C link from the RPi to the Teensy is unreliable. Secondly, the link fails completely when you fire up your RF24L01 radio module. Is that correct?

I've run tests to check the reliability of the I2C connection between a Teensy 4 and a RPI 3A+ using https://github.com/Richard-Gemmell/teensy4_i2c/blob/master/src/tests/raspberry_pi_reliability.cpp. I get roughly 1 error in every 1,000,000 messages with version 1.0.0 of this library. (Earlier versions of teensy4_i2c were much less reliable so it's important to use the latest version.) Here's the code I run on the RPi: [i2c_reliability.zip](https://github.com/Richard-Gemmell/teensy4_i2c/files/4941372/i2c_reliability.zip)

What sort of error rate do you get if you run that with all non-essential wiring disconnected? You might like to set "test_size" to something small to start with as 1,000,000 messages takes a long time.

With regards to the radio module problem, can you confirm that the RF24L01 is connected to the Teensy not the RPi. Please post a link to the RF24 library so I know which one we're talking about.

I haven't used a radio module or an SPI device with my library so I can't guarantee that they'll work. The raw teensy4_i2c driver doesn't depend on timers, it's all interrupt driven. Anything that stops the interrupts being handled in a timely manner will break the I2C link. In practice, that means either doing too much work in an I2C callback or some other interrupt taking too long. 

Sadly, the RPi I2C driver doesn't give very useful error messages. I usually end up debugging things from the Teensy side.

cheers,
Richard

mano1979, 2020-07-18:
I got the connection working reliable now. I switched to software defined i2c pins on the raspberry pi because they allow clock stretching, which is a hardware bug in the i2c hardware of the raspberry pi it seems.

But then my code:

i use this rf24 library for the radio module: https://github.com/nRF24/RF24

The module is indeed connected to the Teensy4.1 and not the raspberry pi.

By the way, sorry for my late reply. By trying to setup the software defined i2c, i managed to brick my pi (software bug in the i2c-gpio.dtbo file) so my pi wouldn't boot anymore and i had to reinstall everything.

mano1979, 2020-07-19:
I think i fixed it all. I still have a few crashes of the program every now and then, but i haven't got pullups installed at the moment (test rig). 
It was never a conflict of the libraries. The moron who wrote part of the code i'm using, gave the same gpio to i2c AND his sensors. Since i don't use those sensors i looked over them en never saw the pins used.

So except for the (probably) pullup issue, all is working. Thank you for your time

(and yes, i realise that i'm a moron too for struggling with this for over a week and not noticing the issue at hand. I noticed it after trying to build up the code from scratch. i then realised that the libraries for i2c and the radio worked together fine.)

Richard-Gemmell, 2020-07-19:
I'm really glad to hear that you're making progress. I've been bitten by bugs like the pin assignment as well. They're only obvious once you've spotted them. :)

Good luck and thanks for letting me know that the libraries work Ok.

I'll leave the issue open for now. Please close it when you're confident that everything is Ok.

cheers,
Richard

closed 2020-08-11 by Richard-Gemmell