Richard-Gemmell/teensy4_i2c #12: Arduino compilation error

pebo111, 2020-11-15:
Hi Richard
To get to know your I2C driver I want to compile the program "find_slaves" and try it on my Teensy 4.0.
But I always get the same error message 
--
C:\Users\admin\AppData\Local\Temp\arduino_build_196192\libraries\i2c_driver_wire\i2c_driver_wire.cpp.o: In function `I2CDriverWire::I2CDriverWire(I2CMaster&, I2CSlave&)':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_driver_wire/i2c_driver_wire.cpp:15: undefined reference to `Slave'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_driver_wire/i2c_driver_wire.cpp:15: undefined reference to `Master'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_driver_wire/i2c_driver_wire.cpp:15: undefined reference to `Slave1'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_driver_wire/i2c_driver_wire.cpp:15: undefined reference to `Slave2'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_driver_wire/i2c_driver_wire.cpp:15: undefined reference to `Master1'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_driver_wire/i2c_driver_wire.cpp:15: undefined reference to `Master2'
collect2.exe: error: ld returned 1 exit status
Bibliothek i2c_driver_wire im Ordner: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\i2c_driver_wire (legacy) wird verwendet
Fehler beim Kompilieren für das Board Teensy 4.0.
-----

What am I doing wrong?
Arduino version 1.8.13
Teensyduino V1.53

With kind regards  Peter

Richard-Gemmell, 2020-11-15:
Hi Peter,
I think this might be something to do with the way you installed the library. I installed it as a "custom library" using the instructions at this link:  https://www.arduino.cc/en/Guide/Libraries#manual-installation

In my case my "Sketchbook location" directory is "C:\Users\Richard\Documents\Arduino\libraries" and I end up with the entire teensy4_i2c source tree in C:\Users\Richard\Documents\Arduino\libraries\teensy4_i2c. It looks like this:
[image]

Once I've done that I can restart Arduino and the examples appear at the bottom of the list in the "File | Examples" drop menu. I can then pick the find_slaves example from the menu like this:

[image]

I can then compile that successfully and run it on my Teensy. I tested this with the same version of Arduino and Teensyduino that you're using.

I hope this helps.
Richard

pebo111, 2020-11-15:
SUPER it works 
Some things are so simple.
Thank you very much for the quick answer.

Many greetings Peter

closed 2020-11-15 by pebo111