Richard-Gemmell/teensy4_i2c #17: Send/Receive more than 256 bytes

cirque-jonb, 2021-05-10:
I'd like to be able to send/receive bytes (under I2C HID protocol) with transaction lengths >= 530 bytes. Was there work or planning done for this feature already?

Richard-Gemmell, 2021-05-12:
I had a quick look at the code. It turns out that it wasn't that difficult to enable larger transfers. Please try the [extend_master_reads](https://github.com/Richard-Gemmell/teensy4_i2c/tree/extend_master_reads) branch.

I've tested it with reads just over 65k. The theoretical limit is 4 Gb. :)

cheers,
Richard

Richard-Gemmell, 2021-05-12:
Here's the [test program](https://github.com/Richard-Gemmell/teensy4_i2c/blob/extend_master_reads/examples/raw/raw_large_transfer/raw_large_transfer.ino) I used.

cirque-jonb, 2021-05-13:
That was fast! Thanks for the help. I'll get to testing it

hmmmmniek, 2021-05-29:
FWIW I also needed this feature! (for reading larger sections from EEPROM) It's working great using the changes from this branch!

Richard-Gemmell, 2021-05-31:
I'll incorporate it into the main branch as soon as I get the test framework sorted out.

PaulN8, 2022-03-26:
I just tried the extended_master_reads branch to go beyond 256 bytes in a single read request and it worked just fine. Definitely a feature worth incorporating into main.