Richard-Gemmell/teensy4_i2c #20: Question about driving multiple I2C slaves from Teensy 4.1

mr-j-tree, 2021-08-18:
Hi Richard!

I have been using the teensy4_i2c library very successfully for a while now, simply to control an I2C buss attached to a Teensy 4 that has just one I2C slave - a Texas Instruments LP series LED driver IC.

I have just produced a new prototype that has a second LED driver IC sharing the same buss (a slightly different model in the same series as the other LED driver), which I'm using to drive some display backlights. I'm experiencing two problems, and I'm having trouble eliminating whether it's something to do with the way I'm using the teensy4_2c library, or whether it's an electrical problem in my hardware design. The I2C buss has a 4.7k pull-up resistor on SDA and another on SCL. The issues are:

1) I2C writes targeting the first chip (the one that was already working with 100% reliability) fail after boot-up about 25% of the time, and I don't know why.

2) Writes to the second chip have never worked. The function `bool write(uint8_t reg, uint8_t value, bool send_stop)` always returns `false`.

So I guess my questions are:

1) Can you confirm that from your experience on a Teensy 4, a single 4.7k resistor is sufficient on an I2C buss containing multiple physical slaves?

2) I have instantiated two seperate I2CDevice objects to control the two different ICs. Would you expect this code to work?:

```
I2CDevice ledDriver = I2CDevice(Master, IC1_ADDRESS, _BIG_ENDIAN);
I2CDevice backlightDriver = I2CDevice(Master, IC2_ADDRESS, _BIG_ENDIAN);
Master.begin(100000);
```

3) Is there any way to get more of a view into what is going wrong that is resulting in the `write()` function returning `false`? Like extract more detail about the error? Essentially I'm trying to determine if the second IC is actually participating on the I2C buss or not.

Cheers and thanks in advance for any help you can offer!

_Josh_

Richard-Gemmell, 2021-08-22:
Hi Josh,

Re: 1) It's more or less impossible to say whether a pullup is big enough or even necessary. It all depends on the wiring. If you've got long wires or a lot of other capacitance you'll need a lower pullup resistance. 

All that really matters is whether the I2C bus contains nice square waves or not. See https://www.i2c-bus.org/i2c-primer/how-i2c-hardware-works/ for more information.

Re: 2) Your code snippet looks Ok. This assumes that IC1_ADDRESS and IC2_ADDRESS are different. :)

Re: 3) Yes. `Master` is an instance of `I2CMaster`. You can call `Master.error()` to get the most recent error code. See the definition of [I2CError](https://github.com/Richard-Gemmell/teensy4_i2c/blob/master/src/i2c_driver.h) for descriptions of the error codes. If you let me know which code you're seeing then I might be able to work out what's happening.

It's possible that your system is suffering from a [stuck bus](https://www.i2c-bus.org/i2c-primer/analysing-obscure-problems/blocked-bus/). It's possible for one I2C device to break the bus why waiting indefinitely for data which never arrives.  The dead give away is that the one of the I2C lines will go LOW permanently. Both lines should be HIGH when the bus is idle. You might be able to see this with a multi-meter.

Some I2C drivers attempt to detect and clear these faults. Unfortunately teensy4_i2c doesn't do this at the moment. (Although it will eventually.) The good news is that I've got another project that can detect and clear stuck bus faults. See https://github.com/Richard-Gemmell/i2c-underneath#bus-monitor.

You can sometimes trigger a stuck bus by rebooting or reprogramming the Teensy while the system is live. In this case you can clear the fault with a cold boot.

It might be worth creating a test rig that contains just a Teensy the 2 LED drivers with absolutely minimal wiring. That way you can test the 2 drivers individually and see if the problems start when using both.

cheers,
Richard

mr-j-tree, 2021-08-22:
Richard!

Wow that's so incredibly helpful. Thanks for taking the time.

I foolishly didn't add any easily accessible I2C test points to my board layout before I had the prototype manufactured, so I'm going to have to get a bit creative with a soldering iron and piggy-back some wires to the teensy's I2C pins to be able to hook it up to a scope and look at the waveform. The LED drivers and traces are extremely small, which is why I had to have the prototype manufactured professionally.

The stuck-buss concept resonates with me... it sounds like this could be the problem, given that the first board fails to do its initial I2C handshake with the LED driver *some* of the time, but rebooting it is also *sometimes* the solution, and then it works with 100% reliability after that.

Thanks - lots to look into there, including the error codes. I'll let you know after a few days what I work out... I know it's always nice to hear back if someone's problem was solved or not!

Cheers,

_Josh_

Richard-Gemmell, 2021-08-22:
The stuck bus thing can be baffling because it can seem so erratic.

FWIW If you've got your whole setup on a single board then you probably don't need pullups. The 22k built into the Teensy may well be enough. The 4.7ks won't hurt though.

If you like to get down and dirty with a scope then you might like to check out the [I2C Specification](https://www.i2c-bus.org/specification/). It's surprising short and clear. It contains the exact electrical requirements for the bus including rise times etc.

If you see anything fishy then please let me know. I haven't got round to doing a deep analysis of the Teensy's output and it's possible that I've got something wrong with the driver setup.

cheers,
Richard

mr-j-tree, 2021-09-20:
Hey Richard!

Just to fill you in on how this has been going. I ended up buying a new Pico Tech USB scope that can capture and decode I2C, in order to be able to see what's going on. The I2C waveforms are interesting... there is definitely a bit of crosstalk between the SDA and SCL lines, in the sense that when the SCL line falls it causes a dip on the SDA, but nothing that takes it out of spec. The link you provided was great, and helped me understand that this is common and normal if it's not too severe.

There's definitely a stuck buss happening, but what's also interesting is that it happens even if I don't boot up the "good" chip at all. There's some sort of issue with the other (newer/untested) chip, so I'm pursuing that in the docs in case there's something I've missed with the logic of the boot-up cycle.

If I still can't lock that down, I think I'll look to replace that chip with a simpler driver that can just take a PWM signal and remove the stuck buss issue all together by just running the one (existing/tested) driver chip on the I2C line. Essentially just taking the shortcut of designing the issue out of the system if I can't solve it with my limited experience.

Thanks again!

Josh

Richard-Gemmell, 2021-09-20:
Thanks for the update Josh. It's really nice to know how things turn out.

Good luck with your project.
Richard

closed 2022-07-03 by Richard-Gemmell