Richard-Gemmell/teensy4_i2c #22: IC2 - Communication between two Teensys 4.1

8JupiterMoll8, 2022-02-20:
Hallo,

I'm triyng to send Data over IC2
from a Teensy 4.1 (Master) >>> to >>> Teensy 4.1 (Slave)

Both Teensys are grounded together and use SCL Pin : 19 and SDA Pin : 18.
The Wires are short.

Can you tell me which code example of your library  i have to use for the master sender and the slave -reciver.

Thx.

8JupiterMoll8, 2022-02-22:
In file included from C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\Wire/Wire.h:26:0,
                 from C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\EasyTransferI2C/EasyTransferI2C.h:45,
                 from src\main.cpp:22:
C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\Wire/WireIMXRT.h:191:16: error: conflicting declaration 'TwoWire Wire'
 extern TwoWire Wire;
                ^
In file included from src\main.cpp:21:0:
lib\teensy4_i2c-master\src/i2c_driver_wire.h:150:22: note: previous declaration as 'I2CDriverWire Wire'
 extern I2CDriverWire Wire;      // Pins 19 and 18; SCL0 and SDA0
                      ^
In file included from C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\Wire/Wire.h:26:0,
                 from C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\EasyTransferI2C/EasyTransferI2C.h:45,
                 from src\main.cpp:22:
C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\Wire/WireIMXRT.h:192:16: error: conflicting declaration 'TwoWire Wire1'        
 extern TwoWire Wire1;
                ^
In file included from src\main.cpp:21:0:
lib\teensy4_i2c-master\src/i2c_driver_wire.h:151:22: note: previous declaration as 'I2CDriverWire Wire1'
 extern I2CDriverWire Wire1;     // Pins 16 and 17; SCL1 and SDA1
                      ^
In file included from C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\Wire/Wire.h:26:0,
                 from C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\EasyTransferI2C/EasyTransferI2C.h:45,
                 from src\main.cpp:22:
C:\Users\Jupiter Moll\.platformio\packages\framework-arduinoteensy\libraries\Wire/WireIMXRT.h:193:16: error: conflicting declaration 'TwoWire Wire2'        
 extern TwoWire Wire2;
                ^
In file included from src\main.cpp:21:0:
lib\teensy4_i2c-master\src/i2c_driver_wire.h:152:22: note: previous declaration as 'I2CDriverWire Wire2'
 extern I2CDriverWire Wire2;     // Pins 24 and 25; SCL2 and SDA2
                      ^
*** [.pio\build\ IC2_SEND\src\main.cpp.o] Error 1
================================================================ [FAILED] Took 3.49 seconds ================================================================The terminal process "C:\Users\Jupiter Moll\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', ' IC2_SEND'" terminated with exit code: 1.

finncat62, 2022-07-02:
Hi, did you get it working? Could you post your basic master writer code, and the slave_reciever(if different from the example) you're using? Thanks!

finncat62, 2022-07-02:
*also, are you using 2.2k 3.3 pullup resistors for both scl and sda, or do you not need them? I'm not sure if the om board small ones are enough. Thanks!

8JupiterMoll8, 2022-07-02:
Hi:

Check this link for SDA and SCL Pinds and Resistor values  : https://www.pjrc.com/teensy/td_libs_Wire.html

Teeny 4.1 RECIVER:
```
#include <i2c_driver.h>
#include <i2c_driver_wire.h>

struct TransmitData
{
  float x;
  float y;
};

TransmitData data;

void setup()
{
  
  Wire.begin();                         // join i2c bus
  Serial.begin(9600);               // start serial for output
  while(!Serial) { }
}

void loop()
{
  Serial.print("requesting ("); 
  Serial.print(sizeof data); 
  Serial.print(" bytes)... ");
  if (Wire.requestFrom(0x40, sizeof data))
  {
    Wire.readBytes((byte*) &data, sizeof data);
    Serial.println("done");

    Serial.println(data.x);
    Serial.println(data.y);

  } 
  else {
    Serial.println("could not connect");
  }
 // delay(500);
}`
```
the other Teensy 4.1 SENDER:

```
#include <i2c_driver.h>
#include <i2c_driver_wire.h>

struct TransData
{
  float x;
  float z;        
};

TransData data;

void requestEvent();

void setup()
{

  Wire.begin(0x40);        // join i2c bus with address #8
  Wire.onRequest(requestEvent); // register event
  //Wire.setClock(1000000);

  Serial.begin(9600);
  while(!Serial) { }

}

void loop()
{
  data.x = 1.0;
  data.y = 2.0;
}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()
void requestEvent()
{
  Serial.print("sending ("); 
  Serial.print(sizeof data);
  Serial.println(" bytes)");

  Wire.write((byte *)&data, sizeof data);
}

```
Good luck...
Greetings from Berlin

finncat62, 2022-07-02:
Hi, thank you very much for the quick response.

I tried the code you gave me and it worked!
However, for my project, the receiver doesn't know when data will come. How can I listen to the stream instead of requesting information?  your github doesn't have a master sender, so I tried to make my own pair based on what you sent me

this is the sender code
#include <i2c_driver.h>
#include <i2c_driver_wire.h>

struct TransData
{
  float x;
  float y;
};

TransData data;

void requestEvent();

void setup()
{

  Wire.begin();

  Serial.begin(9600);
  while(!Serial) { }

}

void loop()
{
  Serial.println("sending");
  data.x = 1.0;

  data.y = 2.0;
  Wire.beginTransmission(9);
  Wire.write((byte *) &data, sizeof data);
  Wire.endTransmission();
  delay(100);

}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()

and this is the reciever code

#include <i2c_driver.h>
#include <i2c_driver_wire.h>

struct TransmitData
{
  float x;
  float y;
};

TransmitData data;
void recieveEvent(int howMany);

void setup()
{

  Wire.begin();
  Wire.onReceive(recieveEvent);                  // join i2c bus
  Serial.begin(9600);               // start serial for output
  while(!Serial) { }
}

void loop(){}

void recieveEvent(int howMany){
  if (Wire.available() > 0)
  {
    Wire.readBytes((byte*) &data, sizeof data);
    Serial.println("done");

    Serial.println(data.x);
    Serial.println(data.y);

  }
}

*the code doesn't have errors, when I copy pasted it into this email it shows up red?[cid:4f2a46a1-26e0-42e6-b356-05ae8393b972]

The website says that for short wire communications (I'm assuming that means the wire is short. Is it something else?) and low speeds the 4.0 on its own should be fine. I disconnected the 2.2kOhm resistors, and it worked!

Thank you so very much,
Greetings from California

ps, in the sender code you sent me line 8 there is a small typo. float z should be float y 🙂
pps, sorry if these are easy questions. I am in highschool and trying to learn as much as I can.

finncat62, 2022-07-02:
Thank you so much!
I plugged the teensies directly into each other, without the pullup pins, used your code, boom works!

Richard-Gemmell, 2022-07-03:
I'm glad to hear you've both got your setups working. Thanks for helping out 8JupiterMoll8.

closed 2022-07-03 by Richard-Gemmell