Richard-Gemmell/teensy4_i2c #24: Slave not receiving 4.0

finncat62, 2022-07-02:
I'm using two Teensy 4.0, on 4.5ish volts and the power doesn't fluxuate.
Pins 18 and 19(sda and scl) are both connected to chained resistors(1000, 1000, 200 ohm) to 3.3v output from the teensy OHM resistors,
The pins also each connect to the corresponding sda scl

This is the slave reciever code
```
// Wire Slave Receiver
// by Nicholas Zambetti <http://www.zambetti.com>
// Modified by Richard Gemmell Oct 2019

// Demonstrates use of the Wire library
// Receives data as an I2C/TWI slave device
// Refer to the "Wire Master Writer" example for use with this
// To use it, connect a master to the Teensy on pins 68 and 17.
//
// Consider using the I2CRegisterSlave class instead of Wire to
// create an I2C slave.

// Created 29 March 2006

// This example code is in the public domain.

#include <Arduino.h>
#include <i2c_driver.h>
#include <i2c_driver_wire.h>

void receiveEvent(int howMany);

int led = LED_BUILTIN;

void setup()
{
  pinMode(led, OUTPUT);
  Wire1.begin(9);                // join i2c bus with address #9
  Wire1.onReceive(receiveEvent); // register event
  Serial.begin(9600);           // start serial for output
}

void loop()
{
  delay(100);
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany)
{
  digitalWrite(led, HIGH);       // briefly flash the LED
  while(Wire1.available() > 0) {  // loop through all but the last
    char c = Wire1.read();
    Serial.println(c);          // print the character
  }          // receive byte as an integer
               // print the integer
  digitalWrite(led, LOW);
}
```

and this is the master sender code 

```

// Wire Master Reader
// by Nicholas Zambetti <http://www.zambetti.com>
// Modified by Richard Gemmell Oct 2019

// Demonstrates use of the Wire library
// Reads data from an I2C/TWI slave device connected to pins 18 and 19.
//
// Consider using the I2CDevice class instead of Wire to read a sensor.

// Created 29 March 2006

// This example code is in the public domain.

#include <Arduino.h>
#include <i2c_driver.h>
#include <i2c_driver_wire.h>

int led = LED_BUILTIN;

void setup()
{
  pinMode(led, OUTPUT);
  Wire1.begin();                         // join i2c bus
  Serial.begin(9600);                    // start serial for output
}
int x = 0;
void loop()
{
  Serial.print("write: ");Serial.println(x);

  digitalWrite(led, HIGH);  // briefly flash the LED
  Wire1.beginTransmission(9);
  Wire1.write(x);   // request 2 bytes from slave device #64
  Wire1.endTransmission();
  digitalWrite(led, LOW);
  x++;
  delay(500);
}
```

Please excuse the poor photos of my circuit.

Power in is usbc, stays at around 4.5-4.7, I feed it to the teensies. They are both on the same ground line, so I believe it should be fine.
The third from top pin on the teensy is 3.3 out, which goes to the first long common line. That is what I put the resistors connected to pins 18 and 19 to. At the teensy after the wire with the resistors, the pins each connect to their own long common line, which then connects to the other teensy. SDA to SDA, SCL to SCL. Sorry for all the white wires, I had only had some leftover scraps to work with.

[image]
[image]

[image]

[image]

[image]

[image]

finncat62, 2022-07-02:
To anyone who sees this, the issue is fixxed. I connected the i2c wires directly to the board without the resistors, and used code the creator sent me:

reciever:```
 
#include <i2c_driver.h>
#include <i2c_driver_wire.h>

struct TransmitData
{
  float x;
  float y;
};

TransmitData data;

void setup()
{
  
  Wire.begin();                         // join i2c bus
  Serial.begin(9600);               // start serial for output
  while(!Serial) { }
}

void loop()
{
  Serial.print("requesting ("); 
  Serial.print(sizeof data); 
  Serial.print(" bytes)... ");
  if (Wire.requestFrom(0x40, sizeof data))
  {
    Wire.readBytes((byte*) &data, sizeof data);
    Serial.println("done");

    Serial.println(data.x);
    Serial.println(data.y);

  } 
  else {
    Serial.println("could not connect");
  }
 // delay(500);
}

```
the other Teensy 4.1 SENDER:
```

#include <i2c_driver.h>
#include <i2c_driver_wire.h>

struct TransData
{
  float x;
  float z;        
};

TransData data;

void requestEvent();

void setup()
{

  Wire.begin(0x40);        // join i2c bus with address #8
  Wire.onRequest(requestEvent); // register event
  //Wire.setClock(1000000);

  Serial.begin(9600);
  while(!Serial) { }

}

void loop()
{
  data.x = 1.0;
  data.y = 2.0;
}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()
void requestEvent()
{
  Serial.print("sending ("); 
  Serial.print(sizeof data);
  Serial.println(" bytes)");

  Wire.write((byte *)&data, sizeof data);
}

```

closed 2022-07-02 by finncat62

finncat62, 2022-07-02:
also, for sender sending whenever and reciever listening, this is code that works for me: 

reciever:
#include <i2c_driver.h>
#include <i2c_driver_wire.h>

struct TransmitData
{
  float x;
  float y;
};

TransmitData data;
void recieveEvent(int howMany);

void setup()
{
  
  Wire.begin(9);       
  Wire.onReceive(recieveEvent);                  // join i2c bus
  Serial.begin(9600);               // start serial for output
  while(!Serial) { }
}

void loop(){}

void recieveEvent(int howMany){
  if (Wire.available() > 0)
  {
    Wire.readBytes((byte*) &data, sizeof data);
    Serial.println("done");

    Serial.println(data.x);
    Serial.println(data.y);

  }
}

sender

#include <i2c_driver.h>
#include <i2c_driver_wire.h>

struct TransData
{
  float x;
  float y;        
};

TransData data;

void requestEvent();

void setup()
{

  Wire.begin();        

  Serial.begin(9600);
  while(!Serial) { }

}

void loop()
{
  Serial.println("sending");
  data.x = 1.0;
  
  data.y = 2.0;
  Wire.beginTransmission(9);
  Wire.write((byte *) &data, sizeof data);
  Wire.endTransmission();
  delay(100);

}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()

Richard-Gemmell, 2022-07-03:
Hi finncat62. 
I'm glad you've solved the problem. Thanks for posting details of your solution. I'm sure it'll help someone else.
cheers,
Richard