Richard-Gemmell/teensy4_i2c #27: Hi, one other quick issue

finncat62, 2022-07-07:
--I don't know where a forum to ask this is, but if you want me to move this there I can --

Hi, I'm using a library called the "Adafruit_PWMServoDriver"
In both the .cpp and .h of the library, I changed the includes to 
(cpp)
Origional:
 #include "Adafruit_PWMServoDriver.h"
#include <Wire.h>
New:
#include "Adafruit_PWMServoDriver.h"
#include "i2c_driver_wire.h"

(.h)
Origional: 
#include <Arduino.h>
#include <Wire.h>

New:
#include "Arduino.h"
#include "i2c_driver_wire.h"

However, when I try to build the code, I get the error message 
Adafruit_PWMServoDriver.cpp.o: In function `Adafruit_PWMServoDriver::Adafruit_PWMServoDriver()':

Adafruit_PWMServoDriver.cpp:40: multiple definition of `Adafruit_PWMServoDriver::Adafruit_PWMServoDriver()'
(Deleted the unnessisry filepath lines)

I tried putting "#pragma once" at the top of the .h, but i got the same error.
What should I do / what am I doing wrong? 

Thank you Mr. Gemmell!

finncat62, 2022-07-07:
turns out it is an issue with the library itself, your code is perfect. Sorry for the worry!

closed 2022-07-07 by finncat62

Richard-Gemmell, 2022-07-11:
No worries. Thanks for letting me know.