Richard-Gemmell/teensy4_i2c #32: two mpu6050 each on a separate i2c bus

tiralonghipol, 2022-08-10:
hi, thank you for this library!
what would be the wiring required to achieve a parallel communication between x2 i2c bus - mpu6050?
using the raw_find_slaves I can only detect address 104 (0x68) but the probelm is that afaik the two mpus have the same address
any idea?
thanks!

Richard-Gemmell, 2022-08-10:
Hi,
The MPU6050 can be configured to use one of 2 different addresses. This is described in section [9.2 I2C Interface](https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6000-Datasheet1.pdf) of the datasheet.

I'm guessing that you're using a breakout board with an MPU6050 on it rather than building your own board. The manufacturer of the board may have given you a way to configure it. On this Adafruit board for example, it's controlled with the [AD0 pin](https://learn.adafruit.com/mpu6050-6-dof-accelerometer-and-gyro/pinouts#other-pins-3044781). It's possible that your particular board doesn't have this option.

The idea is that you leave one board as it is and change the other to use the alternate address which is 105. You can then read both of them with the same I2C port.

On an unrelated note, I suggest you checkout the INT pin if haven't already. You can get dodgy data from an IMU if you read it at the wrong moment!

Hope that helps,
Richard

tiralonghipol, 2022-08-11:
thank you for your fast reply!
what I am trying to achieve, is actually parallel communication, this way:
mpu1 -> bus1
mpu2 -> bus2

I already achieved the communication bus1 -> mpu1/mpu2 using the ad0 pin switch trick
hope is more clear now

Richard-Gemmell, 2022-08-11:
Ah I see.

[This line of raw_find_slaves](https://github.com/Richard-Gemmell/teensy4_i2c/blob/master/examples/raw/raw_find_slaves/raw_find_slaves.ino#L11) determines which I2C bus is being driven by the master. Change it to Master1 to use the second bus. (There's also a Master2 for the 3rd bus.) e.g.
`I2CMaster& master = Master1;`

cheers,
Richard

tiralonghipol, 2022-08-11:
thank you, I will give it a try and let you know!

Richard-Gemmell, 2022-08-11:
FWIW note that [read_async](https://github.com/Richard-Gemmell/teensy4_i2c/blob/master/examples/raw/raw_find_slaves/raw_find_slaves.ino#L32) starts a read by a master. You can start a read on one bus and then immediately start a read on the other one. They'll both run in parallel in the background.

[master.finish](https://github.com/Richard-Gemmell/teensy4_i2c/blob/master/examples/raw/raw_find_slaves/raw_find_slaves.ino#L66) tells you if a read (or write) has finished.

tiralonghipol, 2022-08-11:
that's exactly what I need, thank you again

tiralonghipol, 2022-08-11:
@Richard-Gemmell you are a genius!
as you mentioned in your guide, this was the only change required to my code:

```
// #include <Wire.h>
#include <i2c_driver.h>
#include <i2c_driver_wire.h>
```

on a Teensy 4.0 (600Mhz) I am now able to read both mpu6050 at 1kHz!
without changing any line of code

In case someone else ends up dealing with the same problem, [here](https://github.com/tiralonghipol/arduino/blob/main/sketch_aug10a_many_buses/two_mpus_two_buses.ino) is how I solved the problem (i'm not a software developer, so apologize in advance if I write bad code!)

Richard-Gemmell, 2022-08-11:
Cool. Thanks for letting me know that it works and thanks for posting your code.

I'll close the issue now. Don't hesitate to re-open it if you need.

cheers,
Richard

closed 2022-08-11 by Richard-Gemmell