Richard-Gemmell/teensy4_i2c #34: Error message when installing teensy4_i2c-master

CR2O7, 2023-04-18:
Hi!
When installing the i2C libray teensy4_i2c-master  from a ZIP-file I get the following error messages:
`Invalid version 'v2.0.0-beta.1' for library in: /home/linuxgoran/CloudStation/development/Arduino/libraries/teensy4_i2c-master
Invalid version 'v2.0.0-beta.1' for library in: /home/linuxgoran/CloudStation/development/Arduino/libraries/teensy4_i2c-master
Invalid version 'v2.0.0-beta.1' for library in: /home/linuxgoran/CloudStation/development/Arduino/libraries/teensy4_i2c-master`

Is it something to worry about or just ignore?

I am running Arduino 1.8.19 and Teensyduino 1.58 on Ubuntu 22.04..

Rgds,
Göran

Richard-Gemmell, 2023-04-29:
Hi Göran,

I believe it's a warning from Arduino to say it doesn't like the format of the version number. I think you can ignore it. I'll investigate further.

Does the library work with your setup?

cheers,
Richard

Richard-Gemmell, 2023-04-29:
I don't see this error when installing teensy4_i2c in Arduino 2.0. I'll try 1.8.

Richard-Gemmell, 2023-04-29:
I get this same error as you in Arduino 1.8.19. It's annoying but harmless.
I'll fix it.

Richard-Gemmell, 2023-04-29:
Hi Göran,
I fixed it. The fix is in v2.0.0-beta.2 and tested in both Arduino 1.8.19 and Arduino 2.1.0.
Thanks for reporting this.
cheers,
Richard

closed 2023-04-29 by Richard-Gemmell

CR2O7, 2023-04-30:
Hi Richard,
I'll continue with my project in the autumn, now the boat season is
knocking on the door..
 //Göran