Richard-Gemmell/teensy4_i2c #41: multiple_slave_addresses example crashes

lu-maca, 2025-01-04:
Hi, 

I'm trying to run this example `examples/wire/multiple_slave_addresses/multiple_slave_addresses.ino`, but my Teensy4.1 keeps restarting after calling `beginRange()`. I changed the `setup()` function to:
```cpp

void setup() {
  // Enable the serial port for debugging
  Serial.begin(9600);
  Serial.println("Started");

  // Turn the LED on
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, true);

  // Create a timer to blink the LED
  blink_timer.begin(blink_isr, 500000);

  Serial.println("80");
  // Configure I2C Slave
  wire.onReceive(onReceiveHandler);
  wire.onRequest(onRequestHandler);

  // Listen to every addresses in the range
  wire.beginRange(0x10, 0x20);
  Serial.println("87");
}
```
As you can see, I print the number of lines before I2C slave is configured and at the end of the `setup()` function.

What I get is the following:
```
03:19:33.253 -> Started
03:19:33.253 -> 80
03:19:41.723 -> Started
03:19:41.723 -> 80
03:19:50.306 -> Started
03:19:50.306 -> 80
```

Clearly there is something wrong, but I can't understand what.
Any idea?

Thanks,
Luca

lu-maca, 2025-01-04:
When calling `I2CDriverWire::end()`, the driver tries to access to `master` and `slave` attribute of the class, that do not point to a valid address:
```
13:21:15.745 -> I2CDriverWire::end()
13:21:15.745 -> 0                                      ---- address of master
13:21:15.745 -> 0                                      ---- address of slave
13:21:15.745 -> 536876312                      ---- address of Master1
13:21:15.745 -> 536876516                      ---- address of Slave1
```
`master` and `slave` shall point respectively to `Master1` and `Slave1`

Richard-Gemmell, 2025-01-05:
Hi,
If `master` and `slave` are both 0 (i.e. `null`) then it implies that the instance of `I2CDriverWire` that you're using wasn't initialised. Do you have this line [I2CDriverWire wire = Wire1;](https://github.com/Richard-Gemmell/teensy4_i2c/blob/7cc79e3d09697770c5c2d12ae0e6b8dc6e1d5ee3/examples/wire/multiple_slave_addresses/multiple_slave_addresses.ino#L21) in the same place as the example?

If you're still having problems then please post the simplest `.ino` you can make that illustrates the problem.

cheers,
Richard

Richard-Gemmell, 2025-01-05:
I might have misunderstood. Did `multiple_slave_addresses.ino` failed before you changed it?

lu-maca, 2025-01-05:
Hi @Richard-Gemmell,

thanks for your reply. I'm running the simple example as it is. If I change the line you linked to:
```cpp
[...]
void blink_isr();

// The I2C port we're going to use
// I2CDriverWire wire = Wire1;
I2CDriverWire wire(Master1, Slave1);
[...]
```
it works without any problems.

I'm not sure why this is happening. Can it be a linkage error, maybe? I've never used much external linkage, and I'm not sure how the linker works in these cases.

lu-maca, 2025-01-05:
> I might have misunderstood. Did `multiple_slave_addresses.ino` failed before you changed it?

I'm not sure what you are asking here. Anyway, it failed the first time, as it is. Then I looked into the library and added the `Serial.println()' you se above to understand where it crashes.

Richard-Gemmell, 2025-01-05:
Thanks for letting me know. I notice that I didn't alias wire like this in the other examples so this could be a bug in example. I'll give it a try sometime. I don't have everything set up at the moment so it might be a while before I get round to it.

Thanks for letting me know.

lu-maca, 2025-01-05:
Thank you, I will have a look too, maybe I can understand where the problem is. I'll let you know, in case I find something.

Luca